LabVIEW-DCAF/ModbusModules #33: TCP Master Mode does not set slave device ID

mcornwall, 2017-05-30:
In the Init Case for the TCP Master, it initializes the device with the IP and port but doesn't set the device ID as it is not an input to the init function. Without setting this ID, some TCP Slave devices will not respond. There is an accessor method to set the Device ID. Using this after INIT will allow communication with these devices.

Also of note, the NI Example doesn't do this either!

becega, 2017-05-31:
The standard doesn't use the Device ID for Modbus TCP but it has a Unit Identifier. This field is used for intra-system routing purpose.  It is typically used to communicate to a MODBUS+ or a MODBUS serial line slave through a gateway between an Ethernet TCP-IP network and a MODBUS serial line. This field is set by the MODBUS Client in the request and must be returned with the same value in the response by the server.   
 In the Modbus library it is actually using the Unit ID to build the frame and that is why using the property work.  Most devices use it as 0 by default but it appears some will still use it. We will expose this in the module in the future

smithed, 2017-06-01:
pg 23/46 is the key, here: http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf
Given that 0 = broadcast for serial, I'd suggest defaulting to 1 or 255 (255 is the recommended) and provide a notice message if the user enters a unit ID of 0 to make it clear that, if their device is actually secretly serial, it may not respond.

closed 2017-07-25 by dbendele